maccoylton/esp-homekit-rgbw-strip #1: Using instruction

GPL71, 2019-03-05:
Hello 
I have seen your project
I would like test it.
Waht I need to flash it?
In another thread you'd write the projetct not is reliable.
Why? Has it a bug?

I would like start writing free firmware for homekit and I'm trying understand how to

Thank you

lizzus, 2019-03-06:
You can flash some firmware with OTA using GitHub firmware repository position and main.bin

maccoylton, 2019-03-07:
Hi my module burnt out, not sure if it was something in this code or not. Your mileage may vary. I've added a Wiki page with  instructions for installing

GPL71, 2019-03-07:
Thank you for your replay... @maccoylton 
I have a question
I would like start write free codes but I'm not able to understand steps to prepare ambient on mac and how to generate .bin and other.
Can you help me? My friend @lizzus is interested in it too
Thank you

maccoylton, 2019-03-08:
You will need to be familair with using the command line on MacOS to develop software. 

Follow the instructions here:- https://github.com/SuperHouse/esp-open-rtos to install the SDK. 

Then clone this repository to the top level of you case senssiteive volume and try building it. 

After that you can use this as a pettern to create your own top level driectory to start yuor own project.

maccoylton, 2019-09-02:
Closed out as no further response

closed 2019-09-02 by maccoylton