maccoylton/esp-homekit-rgbw-strip #5: how to install?

toshibochan, 2019-12-31:
i did:
make -C src flash

make: Entering directory '/home/toshi/projects/esp-homekit-rgbw-strip/src'
/home/toshi/projects/esp-open-rtos/common.mk:223: /home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/component.mk: No such file or directory
/home/toshi/projects/esp-open-rtos/common.mk:223: /home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/rgbw_lights/component.mk: No such file or directory
make: *** No rule to make target '/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/rgbw_lights/component.mk'.  Stop.
make: Leaving directory '/home/toshi/projects/esp-homekit-rgbw-strip/src'

what i doing wrong?

maccoylton, 2020-01-01:
Checkout the WIKI, you don't need to build yourself. you can instlal a pre-built formware.

If you do want to build yourself, ensure you execute git submodule update --init --recursive

toshibochan, 2020-01-01:
I was able to install the pre-built but keep disconnecting from WiFi and the dimmer work different from other examples. The problem is the other example the led keep fricling

toshibochan, 2020-01-01:
Your version don’t light is okay but keep disconnecting

toshibochan, 2020-01-01:
And your version the dimming is not smooth

toshibochan, 2020-01-01:
This is the other example 
https://github.com/PA7T/magic_home_strip_rgbw

maccoylton, 2020-01-01:
What do mean by disconnecting? Also what is the issue with dimming? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-01:
I think is Disconnecting from WiFi because in the home app show not responding till I reboot the device. The dimming is not smooth when go up or down.

maccoylton, 2020-01-02:
>I think is Disconnecting from WiFi because in the home app show not responding till I reboot the device. The dimming is not smooth when go up or down. 

@toshibochan try latest release 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-01-04 by toshibochan