maccoylton/esp-homekit-rgbw-strip #6: error to pairing to home app

toshibochan, 2020-01-03:
>>> mdns_reply could not alloc 1800
!!! HomeKit: [Client 5] Failed to compute SRP shared secret (code -2)

maccoylton, 2020-01-03:
I am getting the same issue with new devices .. having Maxim looking at it, will revert when we have a solution

toshibochan, 2020-01-04:
oh okay! please let me know when is fixed.

toshibochan, 2020-01-09:
> I am getting the same issue with new devices .. having Maxim looking at it, will revert when we have a solution

Did you guys find the problem?

toshibochan, 2020-01-09:
@maccoylton you are amazing! I was able to pair with new firmware! Thank you so much! Your firmware is amazing! Have a lot future using EVE app. You are the best!

closed 2020-01-09 by toshibochan

maccoylton, 2020-01-09:
Great, so long as you use version 0.8.2. This issue was running out of memory on pairing. had to re-organise the code a little to not start some of the goodies until after pairing is complete