maccoylton/esp-homekit-rgbw-strip #7: Question

toshibochan, 2020-01-08:
You know if anyone was able to clone the Philips hue led strip firmware? I want be able to use esp to be native suporte to Philips hue

maccoylton, 2020-01-08:
Not sure what you mean? What capabilities are you looking for? On/off change of colour, colour effects? This firmware already has all of those?

toshibochan, 2020-01-08:
I mean, to add the light to hue bridge. I’m looking to buy the hue tv sync but the led strip light are very expensive

maccoylton, 2020-01-08:
OK . so that is outside the scope of what we are trying to achieve here. This is all about creating accessories for use with Apple HomeKiit

closed 2020-01-08 by maccoylton

toshibochan, 2020-01-08:
Hue light is compatible with HomeKit and next hue tv sync going to compatible with HomeKit so in the future I want use esp boards to use with hue

toshibochan, 2020-01-08:
This is the hue sync box but this version is not compatible with HomeKit. Next one will support HomeKit so I’m thinking to buy. This box need hue lights but the led strip from hue is super expensive so I want see if can use ESP8266 to use as hue lights

toshibochan, 2020-01-08:
https://youtu.be/cETauLh0Z00

toshibochan, 2020-01-08:
This talk about new HomeKit hue sync box
https://www.macrumors.com/2020/01/07/philips-hue-new-outdoor-lights/

maccoylton, 2020-01-08:
So you want the Hue Sync box to talk "Hue protocol" to an LED strip controlled power by and ESP8266 or what??

toshibochan, 2020-01-08:
Yes I want use esp8288 led strip to use nativity in the hue app and add to hue sync box

toshibochan, 2020-01-08:
Hue led strip is too expensive so I want use ESP8266

maccoylton, 2020-01-08:
So the ESP8266  is not running HomeKit .. so not part of what I'm interesting in here

toshibochan, 2020-01-08:
No problem thank you @maccoylton