maccoylton/esp-homekit-rgbw-strip #10: white polo issue

toshibochan, 2020-01-10:
I think is bug but when turn on white all RGBW come ON and become cold white instead the only W come ON

toshibochan, 2020-01-10:
my led strip is RGBW is warm white. whit this code (https://github.com/PA7T/magic_home_strip_rgbw) when turn on white only W come on and RGB is off.

maccoylton, 2020-01-10:
Ok . .this is a design consideration, not a bug .. my need was white equals all LEDs come on ie RGB & W.  This primarily to maximize brightness. You can use the temperature option int he Home app to adjust  the warmth. If that does not meet your need the a different implementation would be required,

closed 2020-01-10 by maccoylton

toshibochan, 2020-01-10:
I use the temperature but not get good warm color, still little cold. I wonder if you can add opinion to make only W come on when select white.

maccoylton, 2020-01-10:
I;d recommend creating a separate accessory for that since it is different logic

toshibochan, 2020-01-11:
where in the code I have to change?

maccoylton, 2020-01-11:
>where in the code I have to change?

@toshibochan you would need to decide if you wanted a spectate accessory to show up to control the white leds or what would be the logic of transitioning from colour to white, which would be a complete rewrite of the routine that converts the hsi to rgbw and set the new colour. This is not trivial.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-11:
I want when turn on white the RGB turn off

vniehues, 2020-04-24:
Did you resolve this issue? @maccoylton @toshibochan

toshibochan, 2020-04-24:
@vniehues latest firmware is working great. You can chose if you have RGB or RGBW using eve app.

vniehues, 2020-04-25:
@toshibochan got it, thank you very much!