maccoylton/esp-homekit-rgbw-strip #9: i cant pair more than one controllert

toshibochan, 2020-01-10:
i paired one controller okay but im trying to add other ones and it add but after add the controler is show no response.

toshibochan, 2020-01-10:
I was able to add successful only if changing the name in the pairing process. if the name exist in the home the new device show not respond.

toshibochan, 2020-01-10:
have some bug with naming conflicting.
example:
I can't name bedroom led and living room led because in the bedroom and living room will show only led.

toshibochan, 2020-01-10:
so I named BR Led and LR Led and is working. So strange

maccoylton, 2020-01-10:
>so I named BR Led and LR Led and is working. So strange ...

@toshibochan Max made some updates to the handling of mdns in the latest version of esp-homekit, may be related. If you use a Mac, can you give us the output of the command “dns-sd -B _hap._tcp.”

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-10:
I’m using Ubuntu

maccoylton, 2020-01-10:
avahi-browse is the equivalent on unix, not sure of the full syntax, maybe you an experiment and see what you can find

toshibochan, 2020-01-10:
I want use my Mac but I don’t know how to install in my Mac. You know tutorial how to install on a Mac?

maccoylton, 2020-01-10:
The dns-sd command is run form the terminal and is standard command no installation required.  

I do all my development on Mac. To develop on that, just follow the same setup instructions. What issue did you have?

toshibochan, 2020-01-11:
this?

17:09:46.513  Add        2   6 local.               _hap._tcp.           Led_Strip-RGBW-12760774

toshibochan, 2020-01-11:
even after add to home app give me this log

...........
mDNS announcement: Name=Led_Strip-RGBW-10572192 md=esp-homekit-rgbw-strippv=1.0id=BD:80:88:17:23:EDc#=8002s#=1ff=0sf=0ci=5
     sh=C3eDtA== Port=5556 TTL=4500
on_homekit_event: Server initialised, Free Heap=31952
on_homekit_event: Acessory is paired on initialisation, Free Heap=32432
get_sysparam_info: Sysparam base address 1011712, num_sectors 2
get_sysparam_info: sysparam name: wifi_ssid, value:XXX Airport, key length:9, value length:11
get_sysparam_info: sysparam name: wifi_password, value:**********
get_sysparam_info: sysparam name: ota_repo, value:maccoylton/esp-homekit-rgbw-strip, key length:8, value length:33
get_sysparam_info: sysparam name: ota_file, value:main.bin, key length:8, value length:8
get_sysparam_info: sysparam name: ota_beta, value:n, key length:8, value length:1
get_sysparam_info: sysparam name: ota_version, value:0.8.2, key length:11, value length:5
get_sysparam_info: while loop status 1
accessory_init: GPIOS are set as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: WHITE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: RED GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: GREEN GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: BLUE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: Set gpios as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: Saturation: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Hue: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic

load_characteristic_from_flash: Brightness: int: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: sdk_os_timer_Setfn called

rgbw_set
rgbw_set: Current colour before set r=0,g=0, b=0, w=0,
rgbw_set: led srtip off
rgbw_set:Current colour after set r=0,g=0, b=0, w=0,
>>> HomeKit: Got new client connection: 5 from 10.0.1.35, port 2511
on_homekit_event: Client connected, Free Heap=25460
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 6 from 10.0.1.13, port 5057
on_homekit_event: Client connected, Free Heap=20460
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 7 from 10.0.1.3, port 59370
on_homekit_event: Client connected, Free Heap=18480
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=17792
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: Got new client connection: 8 from 10.0.1.43, port 36304
on_homekit_event: Client connected, Free Heap=16784
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=14636
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: Got new client connection: 9 from 10.0.1.12, port 44499
on_homekit_event: Client connected, Free Heap=13624
>>> HomeKit: [Client 8] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Get Accessories

toshibochan, 2020-01-11:
take about 3 min for the device change from "No Response" be used after add to home app

maccoylton, 2020-01-11:
>this? ...

@toshibochan let it run while you start each of the accessories. This is telling us Led_Strip-RGBW-12760774 is available to homekit and that is the dns name of the accessory. I want to see the second accessory coming online and that it has a unique name.  You can also do dns-sd -L Led_Strip-RGBW-12760774 _hap._tcp. and that will show the details that homekit used to determine what type of accessory it is and if it is paired or not 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

maccoylton, 2020-01-11:
in rountine RBGtoRBGW in colour_conversion.c uncomment the lines and see if that give you the effect you are looking for. 

 /*
        int Bo = (int)(rgbw->blue - Luminance);
        int Ro = (int)(rgbw->red - Luminance);
        int Go = (int)(rgbw->green - Luminance);
 */

toshibochan, 2020-01-11:
I order new controller and arriving today. Later I will test it and I will let you know if work.

toshibochan, 2020-01-12:
i uncomment erasing /* and */ and i get error

int Bo = (int)(rgbw->blue - Luminance);
int Ro = (int)(rgbw->red - Luminance);
int Go = (int)(rgbw->green - Luminance);

make: Entering directory '/home/toshi/projects/esp-homekit-rgbw-strip/src'
CC /home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c: In function 'RBGtoRBGW':
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:128:13: error: redefinition of 'Bo'
         int Bo = (int)(rgbw->blue);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:124:13: note: previous definition of 'Bo' was here
         int Bo = (int)(rgbw->blue - Luminance);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:129:13: error: redefinition of 'Ro'
         int Ro = (int)(rgbw->red);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:125:13: note: previous definition of 'Ro' was here
         int Ro = (int)(rgbw->red - Luminance);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:130:13: error: redefinition of 'Go'
         int Go = (int)(rgbw->green);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/colour_conversion.c:126:13: note: previous definition of 'Go' was here
         int Go = (int)(rgbw->green - Luminance);
             ^
/home/toshi/projects/esp-homekit-rgbw-strip/components/esp-homekit-common-functions/colour_conversion/component.mk:10: recipe for target 'build/colour_conversion//colour_conversion.o' failed
make: *** [build/colour_conversion//colour_conversion.o] Error 1
make: Leaving directory '/home/toshi/projects/esp-homekit-rgbw-strip/src'

toshibochan, 2020-01-12:
i was able to make doing:

        int Bo = (int)(rgbw->blue - Luminance);
        int Ro = (int)(rgbw->red - Luminance);
        int Go = (int)(rgbw->green - Luminance);
/*
        int Bo = (int)(rgbw->blue);
        int Ro = (int)(rgbw->red);
        int Go = (int)(rgbw->green);
*/

toshibochan, 2020-01-12:
@maccoylton you are great! work perfect now only white come on when select white!!!! thanks!!!!

closed 2020-01-12 by toshibochan