maccoylton/esp-homekit-rgbw-strip #11: this supost happen?

toshibochan, 2020-01-12:
time to time this happen in my log. this is okay?

>>> HomeKit: [Client 14] Pair Verify Step 1/2
>>> HomeKit: [Client 14] Pair Verify Step 2/2
>>> HomeKit: [Client 14] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=4668
>>> HomeKit: [Client 14] Verification successful, secure session established
>>> HomeKit: [Client 14] Get Accessories

checkWifiTask WiFi: check interval 200, Status: connection ok: Free Heap=6664, Free Stack=24

maccoylton, 2020-01-12:
Free heap is very low at 6664. Are you using esp-homekit from Max or my version? I found a bug which can cause heap to shrink, fixed in my version but waiting for Max to accept my pull request 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

closed 2020-01-12 by toshibochan

toshibochan, 2020-01-13:
I think I’m using your version because I download your repo and make main.bin

maccoylton, 2020-01-13:
Go into the esp-homekit directory and issue the command 	git remote -v. 

This owl tel you were you pulled that from

toshibochan, 2020-01-13:
git Remote -v
git: 'Remote' is not a git command. See 'git --help'.

The most similar command is
	remote

toshibochan, 2020-01-13:
I think is from Max because in gitmodule say to download Max version

maccoylton, 2020-01-13:
I said to use  git remote -v. You need to type the commands exactly as I recommend

toshibochan, 2020-01-13:
It is max

toshibochan, 2020-01-13:
I will change gitmodule to download your version later

toshibochan, 2020-01-13:
I’m out but I’m access my computer using VNC to check! LOL

toshibochan, 2020-01-13:
My magichome led controller keeping disconnecting from home app and try to pair again. So hope your version will fix this.

maccoylton, 2020-01-13:
ok let me know how you get on

toshibochan, 2020-01-13:
Just change maximkulkin to maccoylton?

[submodule "components/esp-homekit"]
	path = components/esp-homekit
	url = https://github.com/maccoylton/esp-homekit
[submodule "components/esp-wolfssl"]
	path = components/esp-wolfssl
	url = https://github.com/maximkulkin/esp-wolfssl
[submodule "components/esp-cjson"]
	path = components/esp-cjson
	url = https://github.com/maximkulkin/esp-cjson
[submodule "components/esp-ir"]
	path = components/esp-ir
	url = https://github.com/maximkulkin/esp-ir
[submodule "components/esp-homekit-common-functions"]
	path = components/esp-homekit-common-functions
	url = https://github.com/maccoylton/esp-homekit-common-functions
[submodule "components/UDPlogger"]
	path = components/UDPlogger
	url = https://github.com/HomeACcessoryKid/UDPlogger
[submodule "components/esp-wifi-config"]
	path = components/esp-wifi-config
	url = https://github.com/maximkulkin/esp-wifi-config
[submodule "components/esp-adv-button"]
	path = components/esp-adv-button
	url = https://github.com/RavenSystem/esp-adv-button.git