maccoylton/esp-homekit-rgbw-strip #16: clean install and get error to pair to home app

toshibochan, 2020-01-15:
sh=oIBaXg== Port=5556 TTL=4500
on_homekit_event: Server initialised, Free Heap=31928
on_homekit_event: Acessory is NOT paired on initialisation, Free Heap=31928
>>> HomeKit: Got new client connection: 5 from 10.0.1.37, port 52217
on_homekit_event: Client connected, Free Heap=30336
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> HomeKit: Got new client connection: 6 from 10.0.1.37, port 52473
>>> HomeKit: Found older connection with same address as new connection, closing old connection on IP 10.0.1.37, Port 52217, New Port is 52473

>>> HomeKit: [Client 5] Closing client connection
on_homekit_event: Client disconnected, Free Heap=28352
on_homekit_event: Client connected, Free Heap=28348
>>> HomeKit: [Client 6] Pair Setup Step 1/3
>>> HomeKit: [Client 6] Closing client connection
on_homekit_event: Client disconnected, Free Heap=30456

toshibochan, 2020-01-15:
this after pair to home app

                       sh=42MkPw== Port=5556 TTL=4500
on_homekit_event: Server initialised, Free Heap=31928
on_homekit_event: Acessory is NOT paired on initialisation, Free Heap=31928
>>> HomeKit: Got new client connection: 5 from 10.0.1.37, port 49402
on_homekit_event: Client connected, Free Heap=30332
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> HomeKit: [Client 5] Pair Setup Step 2/3
>>> HomeKit: [Client 5] Pair Setup Step 3/3
>>> HomeKit: Added pairing with 13856DE2-8074-4312-BDA3-9D1C52F170A9
on_homekit_event: Pairing added, Free Heap=27380
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Led_Strip-RGBW-10575153 md=model_unknownpv=1.0id=C5:8B:E0:4A:16:5Ac#=1s#=1ff=0sf=0ci=5
                       sh=42MkPw== Port=5556 TTL=4500
>>> HomeKit: [Client 5] Successfully paired
>>> HomeKit: [Client 5] Closing client connection
on_homekit_event: Client disconnected, Free Heap=29948
>>> HomeKit: Got new client connection: 5 from 10.0.1.37, port 49658
on_homekit_event: Client connected, Free Heap=29980
>>> HomeKit: [Client 5] Pair Verify Step 1/2
>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 13856DE2-8074-4312-BDA3-9D1C52F170A9
on_homekit_event: Client verified, Free Heap=28328
on_homekit_event: Acessory is paired on after client validaiton
get_sysparam_info: Sysparam base address 1011712, num_sectors 4
get_sysparam_info: sysparam name: wifi_ssid, value:GTP Airport, key length:9, value length:11
get_sysparam_info: sysparam name: wifi_password, value:**********
get_sysparam_info: while loop status 1
accessory_init: GPIOS are set as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: WHITE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: RED GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: GREEN GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: BLUE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: Set gpios as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: Saturation: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Hue: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Brightness: int: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: sdk_os_timer_Setfn called
rgbw_set
rgbw_set: Current colour before set r=0,g=0, b=0, w=0,
rgbw_set: led srtip off
rgbw_set:Current colour after set r=0,g=0, b=0, w=0,
>>> HomeKit: [Client 5] Verification successful, secure session established
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Pairing added, Free Heap=24540
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
Identify
>>> HomeKit: Got new client connection: 6 from 10.0.1.35, port 31208
on_homekit_event: Client connected, Free Heap=23408
>>> HomeKit: [Client 6] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 7 from 10.0.1.198, port 36824
on_homekit_event: Client connected, Free Heap=19756
>>> HomeKit: [Client 7] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Pair Verify Step 2/2
>>> HomeKit: [Client 6] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=18328
>>> HomeKit: [Client 6] Verification successful, secure session established
>>> HomeKit: Got new client connection: 8 from 10.0.1.167, port 2504
on_homekit_event: Client connected, Free Heap=17308
>>> HomeKit: [Client 7] Pair Verify Step 2/2
>>> HomeKit: [Client 7] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=17080
>>> HomeKit: [Client 7] Verification successful, secure session established
>>> HomeKit: Got new client connection: 9 from 10.0.1.12, port 37847
on_homekit_event: Client connected, Free Heap=16072
>>> HomeKit: [Client 8] Pair Verify Step 1/2
>>> HomeKit: [Client 6] Get Accessories

toshibochan, 2020-01-15:
you can see above i was able to add to home app but keep HomeKit: [Client 8] Pair Verify Step 1/2 and 
HomeKit: [Client 6] Pair Verify Step 2/2 over and over

toshibochan, 2020-01-15:
over and over

!!! HomeKit: [Client 9] Error reading data from socket (code 104). Disconnecting
>>> HomeKit: [Client 9] Closing client connection

toshibochan, 2020-01-15:
have this to 

accessory_init: GPIOS are set as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: WHITE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: RED GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: GREEN GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: BLUE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: Set gpios as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: Saturation: float: 0.000000
load_characteristic_from_flash: Hue: float: 0.000000
load_characteristic_from_flash: Brightness: int: 100
rgbw_lights_init: sdk_os_timer_Setfn called

rgbw_set
rgbw_set: Current colour before set r=0,g=0, b=0, w=0,
rgbw_set: led srtip off
rgbw_set:Current colour after set r=0,g=0, b=0, w=0,

maccoylton, 2020-01-15:
Is this your compiled version or my released bin file? I have 3 running perfectly stable with my released firmware

toshibochan, 2020-01-15:
I did with my and yours and is the same. I not use bin file but I use make -C src flash

maccoylton, 2020-01-15:
Please test using bin file OTA.

toshibochan, 2020-01-15:
you have link of current black_config.bin, otaboot.bin and rboot.bin?

toshibochan, 2020-01-15:
install doing:

esptool -p /dev/ttyUSB0 --baud 115200 write_flash -fs 1MB -fm dout -ff 40m \ 0x0 rboot.bin 0x1000 blank_config.bin 0x2000 otaboot.bin

setup wifi doing:

maccoylton/esp-homekit-rgbw-strip and main.bin

adding to home app but gitting this*

on_homekit_event: Server initialised, Free Heap=31880
on_homekit_event: Acessory is NOT paired on initialisation, Free Heap=31880
>>> HomeKit: Got new client connection: 5 from 10.0.1.37, port 27595
on_homekit_event: Client connected, Free Heap=30300
>>> HomeKit: [Client 5] Pair Setup Step 1/3
>>> HomeKit: Got new client connection: 6 from 10.0.1.37, port 27851
>>> HomeKit: Found older connection with same address as new connection, closing old connection on IP 10.0.1.37, Port 27595, New Port is 27851

>>> HomeKit: [Client 5] Closing client connection
on_homekit_event: Client disconnected, Free Heap=28312
on_homekit_event: Client connected, Free Heap=28312
>>> HomeKit: [Client 6] Pair Setup Step 1/3
>>> HomeKit: [Client 6] Closing client connection
on_homekit_event: Client disconnected, Free Heap=30288

toshibochan, 2020-01-15:
i was able to add and give me thid:

>>> HomeKit: [Client 5] Successfully paired
>>> HomeKit: [Client 5] Closing client connection
on_homekit_event: Client disconnected, Free Heap=30032
>>> HomeKit: Got new client connection: 5 from 10.0.1.37, port 30923
on_homekit_event: Client connected, Free Heap=30280
>>> HomeKit: [Client 5] Pair Verify S>>> HomeKit: [Client 5] Pair Verify Step 2/2
>>> HomeKit: [Client 5] Found pairing with 13856DE2-8074-4312-BDA3-9D1C52F170A9
on_homekit_event: Client verified, Free Heap=28284
on_homekit_event: Acessory is paired on after client validaiton
get_sysparam_info: Sysparam base address 1011712, num_sectors 2
get_sysparam_info: sysparam name: wifi_ssid, value:GTP Airport, key length:9, value length:11
get_sysparam_info: sysparam name: wifi_password, value:**********
get_sysparam_info: sysparam name: ota_repo, value:maccoylton/esp-homekit-rgbw-strip, key length:8, value length:33
get_sysparam_info: sysparam name: ota_file, value:main.bin, key length:8, value length:8
get_sysparam_info: sysparam name: ota_version, value:0.8.4, key length:11, value length:5
get_sysparam_info: while loop status 1
accessory_init: GPIOS are set as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: WHITE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: RED GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: GREEN GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: BLUE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
PWM Stopped 0
rgbw_lights_init: Set gpios as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: Saturation: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Hue: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Brightness: int: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
rgbw_lights_init: sdk_os_timer_Setfn called
>>> HomeKit: [Client 5] Verification successful, secure session established

rgbw_set
rgbw_set: Current colour before set r=0,g=0, b=0, w=0,
rgbw_set: led srtip off
PWM Stopped 45088
rgbw_set:Current colour after set r=0,g=0, b=0, w=0,
>>> HomeKit: [Client 5] Get Accessories
>>> HomeKit: [Client 5] Add Pairing
>>> HomeKit: Added pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Pairing added, Free Heap=24492
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Get Characteristics
>>> HomeKit: [Client 5] Update Characteristics
Identify

toshibochan, 2020-01-15:
installing this way will add to home app with out show "No Response" and was able to use the light right way. but when i make -C src flash "No Rsponse" stay for long time. maybe 10 to 20 min

toshibochan, 2020-01-15:
im using ubuntu to make my main.bin. how i can install using mac?

toshibochan, 2020-01-15:
you can install using make -C src flash to see if is okay?

toshibochan, 2020-01-15:
how i install is. Anything im going wrong?
1-cd projects/
2-git clone https://github.com/pfalcon/esp-open-sdk.git
3-cd esp-open-sdk
4-make
5-export PATH=/home/toshi/projects/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
6-cd ..
7-git clone https://github.com/SuperHouse/esp-open-rtos.git
8-https://github.com/maccoylton/esp-homekit-rgbw-strip.git
9-cd esp-homekit-rgbw-strip
10-git submodule update --init --recursive
11-change the colour_conversion.c
12-export FLASH_SIZE=8
13-export HOMEKIT_SPI_FLASH_BASE_ADDR=0x7a000
14-export FLASH_MODE=dout
15-export SDK_PATH=~/projects/esp-open-rtos
16-make -C src flash

toshibochan, 2020-01-15:
maybe when i change colour_conversion.c is overloading memory?

maccoylton, 2020-01-15:
As above, please test with my .bin file OTA that will help us narrow down possible causes.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-15:
I did test with your main.bin and was okay

toshibochan, 2020-01-16:
i did
make clear all
and now is more stable with my main.bin

toshibochan, 2020-01-16:
i upload my main.bin and main.bin.sig but my sig file is only 4kb and im getting error " error reading flash" when install to magichome. any idea why is only 4kb?

toshibochan, 2020-01-16:
..............
send request......OK
erasing@0x76000>000000...000000 070000...000000 000000...742069 6e6974...104141  so far collected 479232 bytes
send request......OK
erasing@0x77000>414141...00009a  so far collected 480240 bytes

--- ota_verify_hash
--- ota_hash
error reading flash
error reading flash

maccoylton, 2020-01-16:
Ok, that helps. It means either you have the wrong version of one of the components or you changes have in some way exceeded the memory requirement.  I’ll take a look at adding a switch for you.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-16:
Thanks @maccoylton

maccoylton, 2020-01-16:
implemented pure whte on wil give you just w, pure white off will give you rgb+w

closed 2020-01-16 by maccoylton

toshibochan, 2020-01-17:
you are amazing @maccoylton! it work great!!!!!! thank you very much!!!! now work whit rgb and rgbw.