maccoylton/esp-homekit-rgbw-strip #17: Request

toshibochan, 2020-01-18:
Sorry I keep asking things!
Can GPIO0 become push Button?
Single press
Double press
Triple press
And hold reset the magichome

maccoylton, 2020-01-18:
>Sorry I keep asking things! ...

@toshibochan what physical device do you have? None of mine have a physical button?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-18:
Is just the rgbw without IR. I can solder wire to GPIO0 and Ground and add button. Is possible to add code?

toshibochan, 2020-01-26:
I want add push button because sometimes is good if Siri not working and all is good to have option to turn on manually. And can save money because no need to buy extra wireless hue switch

toshibochan, 2020-01-26:
I made this lamp using your main.bin and work great! Is cool if have a button to turn on and off manually to.
[image]

jihao, 2020-03-19:
@toshibochan I did something for my DIY controller which has a button, you may change the pins before use.  

Code here: https://github.com/jihao/esp-homekit-rgbw-strip/tree/dev

maccoylton, 2020-03-27:
If you are happy this works, why don't you submit a push request to add this to the reap

jihao, 2020-03-27:
> If you are happy this works, why don't you submit a push request to add this to this repo

Later, I made some tweak which might not be your intention.  
Like turn on the light by increasing the brightness gradually just like philips hue behavior, but not perfect, it will get a slight flicker at the max brightness ...

So maybe I can extract the button feature to a seperate branch, then submit a pull request.

maccoylton, 2020-03-27:
ok .. no worries

closed 2020-03-27 by maccoylton

toshibochan, 2020-04-21:
hi @jihao you can turn on issue page in your https://github.com/jihao/esp-homekit-rgbw-strip/tree/dev.
the clone or download link is not correct.

toshibochan, 2020-05-06:
Hi @jihao you git clone address not working