maccoylton/esp-homekit-rgbw-strip #18: still having some error when pairing to home app

toshibochan, 2020-01-25:
standard_init:SDK version: 0.9.9, free heap 41360
get_sysparam_info: Sysparam base address 1011712, num_sectors 2
get_sysparam_info: sysparam name: wifi_ssid, value:GTP Airport, key length:9, value length:11
get_sysparam_info: sysparam name: wifi_password, value:**********
get_sysparam_info: sysparam name: ota_repo, value:maccoylton/esp-homekit-rgbw-strip, key length:8, value length:33
get_sysparam_info: sysparam name: ota_file, value:main.bin, key length:8, value length:8
get_sysparam_info: sysparam name: ota_version, value:0.8.5, key length:11, value length:5
get_sysparam_info: while loop status 1
load_characteristic_from_flash: WiFi check interval: int: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
Create accessory name
|manuf='maccoylton' serial='2C:F4:32:B7:E7:89' model='esp-homekit-rgbw-strip' revision='0.8.5' c#=8005
>>> wifi_config: Initializing WiFi config
>>> wifi_config: Connecting to Airport
mode : sta(2c:f4:32:b7:e7:89)
add if0

checkWifiTask WiFi: check interval 200, Status:  default = 1: Free Heap=38372, Free Stack=25
scandone
add 0
aid 1
cnt

connected with Airport, channel 11
dhcp client start...
ip:10.0.1.24,mask:255.255.255.0,gw:10.0.1.1
>>> wifi_config: Connected to WiFi network
on_wifi_ready
get_sysparam_info: Sysparam base address 1011712, num_sectors 2
get_sysparam_info: sysparam name: wifi_ssid, value:Airport, key length:9, value length:11
get_sysparam_info: sysparam name: wifi_password, value:**********
get_sysparam_info: sysparam name: ota_repo, value:maccoylton/esp-homekit-rgbw-strip, key length:8, value length:33
get_sysparam_info: sysparam name: ota_file, value:main.bin, key length:8, value length:8
get_sysparam_info: sysparam name: ota_version, value:0.8.5, key length:11, value length:5
get_sysparam_info: while loop status 1
on_wifi_ready Reset Reason: Soft Reeet
on_wifi_ready: Exception Cause: 0
EPC 1: 0
EPC 2: 0
EPC 3: 0
Exv virtul address: 0
DEPC: 0
Return Address:0
>>> HomeKit: Starting server
>>> HomeKit: Formatting flash at 0x8c000
>>> HomeKit: Generated new accessory ID: 0C:73:50:8C:D8:D2
>>> HomeKit: Generated new accessory key
>>> HomeKit: Configuring mDNS
mDNS announcement: Name=Led_Strip-RGBW-12052361 md=esp-homekit-rgbw-strippv=1.0id=0C:73:50:8C:D8:D2c#=8005s#=1ff=0sf=1ci=5
                                                 sh=pvA4kw== Port=5556 TTL=4500
on_homekit_event: Server initialised, Free Heap=31640
on_homekit_event: Acessory is NOT paired on initialisation, Free Heap=31640
;ld��|�d�|

            �
             d�
               b|����s�b�
                           c��'o�$g'���cp$lbdrl8�o�
                                                       �
                                                         $
                                                          
                                                            
                                                             #g�|��l�
                                                                       c��'o�$��$`�'o
        �dgs�ۓo
                ���dp�g�s�����
                                   
                                    c
                                      '�|䇎x
                                              ��'o��d`�ogl '{���g
                                                                         �� s��'
[... 268 lines cut ...]
>>> HomeKit: Got new client connection: 9 from 10.0.1.167, port 19408
on_homekit_event: Client connected, Free Heap=16972
>>> HomeKit: [Client 9] Pair Verify Step 1/2
>>> HomeKit: Got new client connection: 10 from 10.0.1.35, port 3279
on_homekit_event: Client connected, Free Heap=12332
>>> HomeKit: [Client 10] Pair Verify Step 1/2
>>> HomeKit: [Client 9] Pair Verify Step 2/2
>>> HomeKit: [Client 9] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=11036
>>> HomeKit: [Client 9] Verification successful, secure session established
>>> HomeKit: [Client 10] Pair Verify Step 2/2
>>> HomeKit: [Client 10] Found pairing with 0320352C-C8C2-4DF4-9F99-F57FD1FE9ECD
on_homekit_event: Client verified, Free Heap=12996
>>> HomeKit: [Client 10] Verification successful, secure session established
>>> HomeKit: [Client 9] Get Accessories
>>> HomeKit: [Client 10] Get Accessories
save_characteristics:
save_characteristic_to_flash: On: writing bool value to flash true
save_characteristic_to_flash: Saturation: writing float value to flash 0.000000
save_characteristic_to_flash: Hue: writing float value to flash 0.000000
save_characteristic_to_flash: Brightness: writing int32 value to flash 100
save_characteristic_to_flash: WiFi check interval: writing int32 value to flash 200
save_characteristic_to_flash: Pure White: writing bool value to flash false
>>> HomeKit: [Client 5] Update Characteristics
led_hue_set: timer armed, HUE: 240.000000

maccoylton, 2020-01-26:
What’s the specific issue you are having, the above log looks fine, it the accessory is paired and multiple devices connecting ok. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-01-26:
Have this error:

load_characteristic_from_flash: WHITE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: RED GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: GREEN GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: BLUE GPIO: int8: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
PWM Stopped 0
rgbw_lights_init: Set gpios as follows : W=15, R=12, G=5, B=13
load_characteristic_from_flash: Saturation: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Hue: float: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Brightness: int: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic
load_characteristic_from_flash: Pure White: bool: load_characteristic_from_flash: Error in sysparams error:1 loading characteristic

And

load_characteristic_from_flash: Error in sysparams error:1 loading characteristic

And

HomeKit: Error in geetpeername homekit_server_close_old_duplicate_connection

And

!!! HomeKit: [Client 10] Error reading data from socket (code 128). Disconnecting

maccoylton, 2020-01-26:
The last two are network related .. noting to do with the firmware. 

The others just mean that those parameters have never been changed and therefore written, but this is not an issue since the accessory will be using the defaults. 

What is it about the accessory that is not actually working?

toshibochan, 2020-01-27:
Okay thanks

toshibochan, 2020-01-27:
Is just difficult to pair. Sometimes pair quickly but most the time take many attempts

closed 2020-01-27 by toshibochan

maccoylton, 2020-01-27:
>Is just difficult to pair. Sometimes pair quickly but most the time take many attempts 

@toshibochan this is a known issues with the esp-HomeKit library, various fixes are being tested.  

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>