maccoylton/esp-homekit-rgbw-strip #20: having issues installing from OTA

toshibochan, 2020-02-04:
having issues installing from OTA. I set to maccoylton/esp-homekit-rgbw-strip and is trying to download version 0.0.0.

[image]

maccoylton, 2020-02-04:
>having issues installing from OTA. I set to maccoylton/esp-homekit-rgbw-strip and is trying to download version 0.0.0. ...

@toshibochan this look like an issue accessing GitHub. I would recommend you post this into LCM GitHub repository. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

maccoylton, 2020-03-27:
Did you ever get this resolved?

toshibochan, 2020-03-28:
Been a while I try I will look later.

toshibochan, 2020-04-05:
this link otaboot.bin from wiki not good. (https://github.com/HomeACcessoryKid/ota/releases/)
I download otaboot.bin from this and work okay (https://github.com/HomeACcessoryKid/life-cycle-manager/releases/)

maccoylton, 2020-04-07:
Thanks I've update the wiki

closed 2020-04-07 by maccoylton

toshibochan, 2020-04-07:
Did you update the wiki? The wiki look like is the same.

maccoylton, 2020-04-08:
Yes .. section 2 is quite different :- 

To install the OTA firmware we need to download the following:- bootloader.bin file from here:- https://github.com/maccoylton/esp-homekit-rgbw-strip/release

You then need to erase and flash these three files to your device as follows, I assume here you have esptool.py installed and know how to put you device into flash mode:-

esptool.py -p /dev/cu.usbserial erase_flash esptool.py -p /dev/cu.usbserial erase_flash esptool.py -p /dev/cu.usbserial --baud 230400 write_flash -fs 8m -fm dout -ff 40m 0x0 bootloader.bin

toshibochan, 2020-04-08:
The link is not working. Is missing “s” on release”s”

https://github.com/maccoylton/esp-homekit-rgbw-strip/release

maccoylton, 2020-04-08:
Fixed.. thanks