maccoylton/esp-homekit-rgbw-strip #22: update not working

toshibochan, 2020-05-20:
im trying to update to 0.8.6 from 0.8.5 but not working. one was possible to update one device but 4 other device not updating and showing 0.8.5 and one now showing firmware 0.0

maccoylton, 2020-05-21:
>im trying to update to 0.8.6 from 0.8.5 but not working. one was possible to update one device but 4 other device not updating and showing 0.8.5 and one now showing firmware 0.0

@toshibochan what version of OTA firmware are those devices running. Please remember that OTA version 1  was broken by a change made by github. The only fix for that is to flash the new OTA version 2 over a serial interface. Version 2 has a number of options to avoid that situation In future. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-05-22:
maybe was v1. I will reinstall again later

closed 2020-05-22 by toshibochan