maccoylton/esp-homekit-rgbw-strip #23: issue when lose internet connection

toshibochan, 2020-05-20:
issue when lose internet connection. when the device lose internet connection the led turn red and soon after the light turn off.

maccoylton, 2020-05-21:
That very strange, I’ve never seen that before. I assume a restart fixed that? 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-05-22:
Maybe is the same as sonoff-basic esp-open-rtos problem?

maccoylton, 2020-05-22:
It is, the strange pice is why the light turns RED. What GPIOs does your module use?

toshibochan, 2020-05-22:
[image]

maccoylton, 2020-06-11:
I'm sorry I have not idea what this would happen, Can you provide a log file using udplog-client.

closed 2020-08-10 by toshibochan