maccoylton/esp-homekit-rgbw-strip #25: today all my led controllers disconnect from HomeKit

toshibochan, 2020-06-02:
today iOS and TV OS and HomePod os was updated and later tonight all my led controllers keeping disconnecting from HomeKit and rebooting. Some forgot the GPIO setting when come back to homeckit. Anyone have this problem.

maccoylton, 2020-06-03:
Hi I haven't seen that issue. How many accessories do you run on your network and how many controllers?

toshibochan, 2020-06-03:
9 magic home led controller, 5 son off mini 8 son off basic

toshibochan, 2020-06-03:
maybe because I'm using family sharing and I'm the guest and im the one adding the device?

maccoylton, 2020-06-05:
How many Mac's/IOS devices using homekit ?

toshibochan, 2020-06-05:
3 Macs, 2 iPhones 2 iPads

maccoylton, 2020-06-05:
>3 Macs, 2 iPhones 2 iPads 

@toshibochan I think we are the limits of memory, too many controllers (Mac’s or IOS devices) can take us over the edge. So with the update and device re-pairing that may tips is over the edge. 

There is some work ongoing to maximize memory utilization. The other option is to limit the number of controllers 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-06-05:
You mean memory in the controller?

maccoylton, 2020-06-05:
no the accessories.. .. each controller that connects to an accessory consumes some memory on the accessory

closed 2020-06-11 by maccoylton