maccoylton/esp-homekit-rgbw-strip #27: 0.9 firmware "Couldn't add Led_Strip-RGBW"

toshibochan, 2020-08-09:
I just did clean install on one controller and I'm trying to add to HomeKit but have error.

Couldn't add Led_Strip-RGBW

[image]
[image]

maccoylton, 2020-08-09:
Sorry the new version of esp-homekit has fundamentally changed the way memory is managed, it preallocates a lot of stuff, leaving less memory for the accessory. In my case I just update, But in your case sine you reinitialized , and the paring process uses a lot of memory, then it is having an issue.  Reverting to a previous version is the only option till we get to a fix.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-08-10:
how to install previous version?

toshibochan, 2020-08-10:
0.9.1 is good to add to HomeKit. Was very fast to pair. Now I will test if will lose connection.

closed 2020-08-10 by toshibochan