maccoylton/esp-homekit-rgbw-strip #28: strange behavior in HomeKit after I add your esp-homekit-rgbw-strip

toshibochan, 2020-08-11:
2 weeks now I stopped using your firmware and start using different firmware to all my magic home and sonoff devices. This pass 2 weeks HomeKit was working good. Today I try only one magic home 0.9.1 and pear quick to home app but problem starts. Many devices start showing "Not Response" after I add only this one device. And I can't control in home app and homepod. After unpear the magic home everything start working again. Is so strange one device break other devices. Something in your firmware is making other to show "No Response".

maccoylton, 2020-08-12:
> 2 weeks now I stopped using your firmware and start using different firmware to all my magic home and sonoff devices. This pass 2 weeks HomeKit was working good. Today I try only one magic home [...]

@toshibochan Max is making significant changes to the esp-HomeKit library and this is taking some time to stabilse. I’d advise if you want to use my firmware you revert to the earlier version which was more stable. 

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

toshibochan, 2020-08-13:
Is okay I’m only trying one extra test device. I will try all the New versions. When finally get stable I will install in all device. Was just strange one device affected all other devices in homekit.